UNCG-DAISY/Instagrain #7: determine best distance from the bed for camera

ebgoldstein, 2021-05-17:
this will be determined via back and forth with the seidnet model

ebgoldstein, 2021-05-17:
from @jacobstasiewicz:
I took 3 pictures of a camera calibration board from 3 different distances to calculate pixel per inch. The results are as follows:
distance: 4 7/8 has a PPI of 624.38
distance: 3 1/2 in has a PPI of 858.52
distance: 2 1/2 in has a PPI of 1144.69

jacobstasiewicz, 2021-05-19:
Ordered tripod to take precise pictures at different elevations and run through SediNet

jacobstasiewicz, 2021-05-25:
Took 15 pictures at varying heights from 3 inches to 7 inches above the bed to run through sedinet model

jacobstasiewicz, 2021-05-25:
Cropped all pictures to 512 x 512 for sedinet model

jacobstasiewicz, 2021-05-26:
did calculations on lens angle to find pixels per mm

jacobstasiewicz, 2021-06-29:
found to be 3in

closed 2021-06-29 by jacobstasiewicz