UNCG-DAISY/Instagrain #9: boot to camera

ebgoldstein, 2021-05-17:
write a script so that, upon startup, open the camera.

put scripts in the public github repository (`/camerasoftware` directory).

jacobstasiewicz, 2021-06-04:
Somewhat got it to work, still having issues on pulling up terminal to specify folder

ebgoldstein, 2021-10-07:
@jacobstasiewicz — are you ok if i write some code (via `cron`) so that the shell script runs as soon as the camera is booted?

jacobstasiewicz, 2021-10-07:
Yes I am, although I couldn't get the script to run via cron

ebgoldstein, 2021-10-07:
i will try, and likely fail...

ebgoldstein, 2021-10-14:
I did not try yet. in many ways the its fine at this stage if the users need to double tap the shell script to make the camera run.. i say this because debugigng is easier if the script does not run immediately