UNCG-DAISY/Instagrain #19: Contact Dan Buscombe

jacobstasiewicz, 2021-05-19:
to discuss lighting options for camera and how to format spreadsheet for SediNet

ebgoldstein, 2021-05-21:
LED rings as far back as possible. very diffused light. as uniform as possible 

avoid specular reflections with light too close to the lens, and internal reflections.

ebgoldstein, 2021-05-21:
the P is Pixel, D is grain size...  so D50 is grain size in mm P50 is grain size in pixels... to convert just do mm/pixels..

jacobstasiewicz, 2021-05-21:
So how would I convert our sieved samples to grain size in pixels? I am a little confused now

ebgoldstein, 2021-05-21:
The idea is that each of these images from sedinet is from a sieve analysis, grains that make it through the upper sieve, but not the lower sieve. Each sample has a range of grain sizes. D5, D10, D50, etc. 

Images of these samples are taken, and the pixel dimension are determined.. so X mm/ pixel.. Then all teh grain size metrics are converted to Pixel Size metrics (P5, P10, P50, etc.)...  by dividing grain size metric (mm) by known pixel size (mm/pixel). 

that is what is expressed in these CSVs... does that make sense? so the steps for us would be:

- determine grain size metrics for a given sand sample
- take a picture of the sample
- compute pixel size of image
- convert the grain sizes metrics for the sample in 'P' metrics by dividing by the  pixel size.

closed 2021-05-24 by ebgoldstein

jacobstasiewicz, 2021-05-24:
Okay, I think I might of sent Dan an email before I saw you posted this but it clears everything up. So after I have done all of the pictures, computed pixel size and converted metrics to pixel size, I should be able to run it through the SediNet