UNCG-DAISY/Instagrain #38: GPS does not always work in the field...

ebgoldstein, 2021-10-13:
The GPS does not always provide a location, and i was having some issues with exceptions..

I adjusted the script slightly so -9999 is the default and new values write on top (i will link my commit to this issue), but i am not sure that `gps.next()` is the best way to make sure a position is returned... 
i know this is an old S.O. thread, but i wonder if it hints at a way to deal with it... https://stackoverflow.com/questions/6146131/python-gps-module-reading-latest-gps-data

jacobstasiewicz, 2021-10-18:
Ill give this a look when I get the camera back

ebgoldstein, 2022-02-03:
we are switching the GPS to https://www.adafruit.com/product/2324, 

which has a tutorial here: https://learn.adafruit.com/adafruit-ultimate-gps-hat-for-raspberry-pi/

jacobstasiewicz, 2022-03-10:
GPS working reliably now with gpsd package. Tested with and without fix