UNCG-DAISY/Instagrain #70: SD card image.

ebgoldstein, 2022-01-24:
probably the easiest way to reproducibly build/package the camera software is to provide an SD card image.. Then users could just have all the dependencies installed -i.e., they would just use balena etcher and everything would just work on the pi when they inserted the sd card..

 we could save the image on a zenodo repository, and link it here..

https://www.tomshardware.com/amp/how-to/back-up-raspberry-pi-as-disk-image

jacobstasiewicz, 2022-01-24:
Hey Evan I agree, good idea. After we speak tomorrow I can get on that

jacobstasiewicz, 2022-02-02:
Imaged a new sd card succesfully, both work. Now I am working on creating an image to update to a zenodo repo

jacobstasiewicz, 2022-02-15:
Uploaded to syno drive at https://cshl.synology.me:5001/d/f/670216654824718296

closed 2022-02-15 by jacobstasiewicz