UNCG-DAISY/Instagrain #90: PDF of grain size

ebgoldstein, 2022-06-03:
it might be helpful to display a PDF of the grain size, and not just the CDF. (this suggestion is from @ncohn).

I am not sure how to display it graphically (if they should be on the same plot panel? or with a button? or two panels?)

here is a code snippet for the needed steps w/ `scipy.misc.derivative`:

https://datascience.oneoffcoder.com/pdf-cdf.html

jacobstasiewicz, 2022-06-08:
evan I think it would be easiest to make the entire plot a button, so when clicked it will switch between the two types

ebgoldstein, 2022-06-09:
Good idea

ebgoldstein, 2022-06-18:
@jacobstasiewicz - give this a test on the device. There was no plot showing when i tried it this morning on my device

jacobstasiewicz, 2022-06-18:
give it a try now

ebgoldstein, 2022-06-18:
works - super nice!.. the code also seems faster to me!

closed 2022-06-18 by ebgoldstein

jacobstasiewicz, 2022-06-18:
Thanks! Is there a way to quantify the speed of the code? I would like to run a test and make a base line so when we are streamlining the code I can see what methods work best, just out of curiosity

ebgoldstein, 2022-06-18:
i made a new issue for that..