UNCG-DAISY/Instagrain #91: RTC woes..

ebgoldstein, 2022-06-06:
I think the instructions for the RTC are not correct.. i can work on this

ebgoldstein, 2022-06-09:
https://forums.adafruit.com/viewtopic.php?f=50&t=182292&p=885900&hilit=gps+hat+rtc#p885900

jacobstasiewicz, 2022-06-13:
The csv file is now updated from the gps in UTC, but it would still be nice to figure out the RTC

ebgoldstein, 2022-06-13:
let's keep this open, and i can rename this issue

ebgoldstein, 2022-06-20:
new rtc has been ordered.. this would mean swapping out {GPS hat+connector+antenna} for {RTC + USB-GPS).. this would require a bigger hole on the top case (for USB cable).

ebgoldstein, 2022-08-24:
I think this is not needed anymore with the RTC in V3, the GPS time recorded in the sample spreadsheet, and the 5 digit hash code being used for the folder name

closed 2022-10-12 by ebgoldstein