UNCG-DAISY/Instagrain #93: metadata and naming

ebgoldstein, 2022-06-06:
The camera needs some way to record metadata and/or naming of the picture for each session beyond 1,2,3,... 

This is a discussion, and i have some ideas..

jacobstasiewicz, 2022-06-18:
can you explain a little more how I would do this with a hash map, I am currently reading this website https://medium.com/eonian-technologies/file-name-hashing-creating-a-hashed-directory-structure-eabb03aa4091,

ebgoldstein, 2022-06-19:
I'm thinking simpler, where each picture can be 1,2,3,4,.....
but the folder is a random string of letters... and users would then write down the string in their notebook as a key to the location/time/experiment...

(this is a half-way step to users being able to input folder names...)

here is my solution:

```
import random, string

def random_five():
    return ''.join(random.sample(string.ascii_uppercase,5))

FOLDER_NAME = random_five()

```

from https://stackoverflow.com/questions/2823316/generate-a-random-letter-in-python

what do you think @jacobstasiewicz ?

ebgoldstein, 2022-06-20:
this works for me, and i implemented it via ... https://github.com/UNCG-DAISY/SandCam/commit/8af79891deb38718c957fa1bc4e7eb2a9b2389f7

https://github.com/UNCG-DAISY/SandCam/blob/8af79891deb38718c957fa1bc4e7eb2a9b2389f7/software/main_tk.py#L58-L67

to turn it off, you can comment lines 61-63 and 67 and  uncomment line 66

closed 2022-06-20 by ebgoldstein