UNCG-DAISY/Instagrain #95: software version #

RyanMieras, 2022-06-10:
1. csv file (or separate metadata file?) should include ascii version of software number (and machine learning model too?)
2. Need to shift over to tagged releases

jacobstasiewicz, 2022-06-17:
Could this be done in the first line of the code? also what is a tagged realese?

RyanMieras, 2022-06-17:
Tagged code is a numbered version, like 2.3.1, for example, which is fully packaged up and is an unedited version beyond that state. There are plenty of videos on yt explaining in more detail, like this one: https://youtu.be/vSsypsDRiMU

If you want to try it, I'd recommend starting a separate git repo and put fake test code in there. Then tag a version 1.0. Then update your main "working repository" is what it's called, and edit some files. Then tag the updated working copy as version 1.1.

Get it to work in a test repo before you mess around with it too much in an actual working copy of the SandCam repo.

ebgoldstein, 2022-06-17:
I can handle the first release, @jacobstasiewicz you can watch while i do it. I will aim for sometime next week.
Before a release, i need to:
- [ ] finish #98 
- [ ]  Make a model repository on Zenodo, and put the tflite models on there.

@jacobstasiewicz , can you: 
- [ ] add two variables at the top of the code to define `software_version` and  `model-version`(These values can both be set to '0.1' for now)
- [ ] add lines in the code that write these two variables to the csv

jacobstasiewicz, 2022-06-18:
yes

ebgoldstein, 2022-06-20:
models can stay in the repo for now, no need to use zenodo..

ebgoldstein, 2022-06-20:
here is the tagged release: https://github.com/UNCG-DAISY/SandCam/releases/tag/0.1

closed 2022-06-20 by ebgoldstein