UNCG-DAISY/Instagrain #99: speed test on-device code

ebgoldstein, 2022-06-18:
as mentioned at the end of #90, we should get the speed of each of the components of the code on the device. 
the simple way is with well placed `time.time()` commands.. i.e., https://stackoverflow.com/questions/5849800/what-is-the-python-equivalent-of-matlabs-tic-and-toc-functions#5849861

it would also be cool to do some profiling of the code, but that might require some other tooling..

jacobstasiewicz, 2022-06-23:
Below is the time it takes for the program to start up and create the tk window 

importing packages : 3.5500845909118652
defining pins and vars: 0.8810551166534424
GPS connect: 0.0010991096496582031
TFLITE_stuff: 2.823082447052002
Create_direcs: 0.0024917125701904297
Create textfile: 0.0006253719329833984
Create functions: 3.910064697265625e-05
make tk widgets function: 0.3118555545806885

Below is the time for each function
capture function: 2.791738748550415
sed_stats function: 0.0011866092681884766
coord_update function: 0.04084444046020508
photo_update function: 0.6022562980651855
make_plot function: 0.8402371406555176
plot_update function: 0.021099567413330078
updategui function: 1.5102274417877197
capturegui function: 4.72385048866272

ebgoldstein, 2022-06-23:
This is good to know