UNCG-DAISY/Instagrain #114: Launch program  at start-up

dbuscombe-usgs, 2022-08-15:
Run.sh can be very hard to launch in the field, eg when wet or sandy. It can take 10 or more tries to launch by double click. I'd prefer the program launched at startup

ebgoldstein, 2022-08-16:
yes, this was an early issue #9 which we decided to close b/c it was annoying.. but its worth figuring out now..

jacobstasiewicz, 2022-08-25:
Worked on this for a while, should be as simple as adding a single line to crontab but it is not. Do you have any ideas on this?

ebgoldstein, 2022-08-25:
@dbuscombe-usgs - ideas?

ebgoldstein, 2022-08-25:
I will also look into this..

ebgoldstein, 2022-08-25:
ok, i think its a sudo thing, and how the python packages are installed...

ebgoldstein, 2022-08-25:
I have tried.. i am trying rc.local.. but it is not working... I found this: https://learn.sparkfun.com/tutorials/how-to-run-a-raspberry-pi-program-on-startup/all

which suggests the reason its not working is that i am trying to use graphical elements before its possible...

ebgoldstein, 2023-05-30:
this is now done via instructions here: https://learn.sparkfun.com/tutorials/how-to-run-a-raspberry-pi-program-on-startup/all

ebgoldstein, 2023-05-30:
i.e., in terminal type:

`mkdir /home/pi/.config/autostart`
`nano /home/pi/.config/autostart/instagrain.desktop`

in nano, type this:
```
[Desktop Entry]
Type=Application
Name=Instagrain
Exec=/home/pi/Desktop/RUN.sh
```
then control-x to save.. leave nano and reboot..

ebgoldstein, 2023-05-30:
there might also be other options, listed in: https://raspberrypi.stackexchange.com/questions/98944/launch-a-gui-tkinter-program-on-boot