UNCG-DAISY/Instagrain #115: Increment sample numbers better

dbuscombe-usgs, 2022-08-15:
Sample number goes back to 1 on power up. When sampling in the field, system is sometimes powered off to save battery. Sample numbering system independent of power cycling would be preferable for easier note taking and match-up with physical samples

ebgoldstein, 2022-08-16:
yes, we have discussed this in #96, as well as #102 ..

Essentially, a user needs to be able to reoccupy a folder.. this might be a text box before the program is run?

We also need documentation on this

jacobstasiewicz, 2022-08-26:
fixed with recent commit

ebgoldstein, 2022-08-26:
this is not quite working on my device. I can debug it...

ebgoldstein, 2022-08-31:
ok, i found the issue @jacobstasiewicz : in directories where no pictures exist, there is no `last_line`, so the program fails. 

This is really a corner case - operators should just start a new session/directory in this situation, but its good to know about it... I am not going to write any modifications now, but i will add documentation telling people not to reoccupy empty directories

https://github.com/UNCG-DAISY/Instagrain/blob/21aa04fdc7d24d89b157d10dcb4dd433adad4138/software/main_tk.py#L448-L452

closed 2022-08-31 by ebgoldstein