UNCG-DAISY/Instagrain #118: While building V3, rewrite v3 instructions and BOM based on changes

ebgoldstein, 2022-08-19:
Notes so far:
- no coral
- USB GPS
- RTC

ebgoldstein, 2022-08-25:
this issue should close with a tagged release (v0.2) and likely also a link to a RPi OS image on zenodo

ebgoldstein, 2022-08-25:
this also might result in upgrading the software to the new RPi OS and the new camera software

jacobstasiewicz, 2022-09-01:
software upgrade is going to be needed for arducam

ebgoldstein, 2022-12-07:
basic instructions are up... mostly copied from v2

ebgoldstein, 2023-03-14:
this should reflect v3 w/ HW cam and v3 w/ autofocus camera

ebgoldstein, 2023-03-21:
ok, i have updated the GH repo and the wiki to reflect:

v1 is 3d printed
v2 is large enclosure
v3 is small enclosure

Software is essentially identical for all of these versions

There are now 2 folders for v3 software: ` v3_HQ` and `v3_auto` - these are substantially different, based on the HQ nad autofocus cameras..

closed 2023-03-21 by ebgoldstein