UNCG-DAISY/Instagrain #135: Parts for dev work

ebgoldstein, 2022-09-20:
In the course of discussing a 'dev camera', it turns out we just need a simple way to prototype cameras and do development work that do not rely on having an actual enclosure...

the RPi can be attached to the RPI screen (with the included standoffs).. and then we just need like a plate that can attach to the hood (w/ peg holes) and then holes for the camera standoff stuff...

_Originally posted by @ebgoldstein in https://github.com/UNCG-DAISY/Instagrain/issues/133#issuecomment-1252577400_

ebgoldstein, 2022-09-21:
via @jacobstasiewicz 

"has raspberry pi on left, battery slots on right, button on top, camera mounting holes, and the peg holes for the hood.. little 5mm  bumps on the bottom for when the hood or camera are not on"

[image]
[image]

jacobstasiewicz, 2022-10-05:
Added file to github

closed 2022-10-05 by jacobstasiewicz