UNCG-DAISY/Instagrain #137: New RPi image

ebgoldstein, 2022-09-26:
Make a new RPi image for v2 camera, with new model.. might require a pip udpate and a new TF interpreter version

ebgoldstein, 2022-09-27:
https://www.tomshardware.com/how-to/back-up-raspberry-pi-as-disk-image

https://magpi.raspberrypi.com/articles/back-up-raspberry-pi

ebgoldstein, 2022-09-27:
this is done.. still thinking about how to shrink it and serve it

ebgoldstein, 2022-09-27:
the image is 32 GB. A `.tar.xz` is 1.2GB..

ebgoldstein, 2022-09-27:
I tested writing the image on a 32GB SD card using the Rsaaspberry Pi imager.  It worked (:tada: ).
@jacobstasiewicz, i will send you a link to the image, and ask that you test it too..

ebgoldstein, 2022-09-29:
ok, it seems to be working.. I need to put it on Zenodo and then close this issue...

WFarebrother, 2022-11-14:
@ebgoldstein If possible, can I please get a copy of the image to test? Currently attempting to put a V2 device together. Thanks!

ebgoldstein, 2022-11-14:
@WFarebrother - here is a google drive link for the v2 image: https://drive.google.com/file/d/1Vkpmd04zKXRPN8tuaD6aKBoAVZ30dl68/view?usp=sharing

please let me know how you get on! and feel free to reach out if you want to discuss further!

(I am going to close this issue for now, and wait for a full zenodo release until after some testing, etc.....)

closed 2022-11-14 by ebgoldstein